SUWS/mast-planning #11: Plan framework for cable mounting

Tyler-Ward, 2015-07-12:
###### Dependencies: 
- [ ] knowing the lengths and types of cable. #8 

###### Sub tasks
- [ ] Design and source parts for junction box.
- [ ] Build junction box.
- [ ] Determine how cables should be mounted along the mast / crossbeam and against the wall of SUSU.

Tyler-Ward, 2015-07-12:
Potential plan from wiki, probably wants revising

* Connection plate
   * ?* weatherproof N through connectors [[17]]
   * ?* N blanking plates
* Mounting frame (suggest stainless or aluminium)
* Connector box [[18]]
   * 2* ethercon panel connectors [[19]]
   * 2* XRL7 male panel [[20]]
   * 2* XLR7 female panel [[21]]
   * 2* ethercon cable [[22]]
   * 2* XLR7 female cable [[23]]
   * 2* XLR7 male cable [[24]]
   * potting compound [[25]] / lots of silica gell
   * D connector mounting plate [[26]]
   * weatherproofing ans sealant [[27]]

Tyler-Ward, 2015-07-21:
@philcrump do you think we should have the cable connection plate vertical or horizontal (as in the cable passed through from top to bottom or from left to right)?

Tyler-Ward, 2015-07-26:
Here is a design of an RF connection plate (with connectors on both sides but no cables as yet)
[image]

drn05r, 2015-08-02:
When cabling is vertical underneath the mast the mounting clips look to be screwed in presumably with rawlplugs inserted in drilled holes.  The horizontal cabling to the point the cables enter the building look to be simple nailed in mounting clips.

Tyler-Ward, 2015-08-07:
A new plan will be needed based on the results of #44.