SUWS/mast-planning #20: Design and build improved water-proofing for Wi-Fi dish

Tyler-Ward, 2015-07-12:
###### Sub tasks
- [ ] Design improved water-proofing
- [ ] Source any parts consumables need to build and attach housing to dish
- [ ] Build water-proofing

drn05r, 2015-08-02:
We shall use waterproof sealant (i.e. that used round the edge or baths) to see round the clip-on cover over the Ethernet port, including drain hole, after putting silica bags inside.  We will also put sealant between the central column and the dish itself, so if water collects when the dish is pointing directly up it cannot get into any electronic parts.

Tyler-Ward, 2015-08-04:
http://www.screwfix.com/p/no-nonsense-general-purpose-silicone-clear-310ml/35887?cm_sp=Search-_-SearchRec-_-Area1&_requestid=279618#_=p as an option for waterproofing