SUWS/mast-planning #27: Get SUSU to instal server rack in the "playroom"

Tyler-Ward, 2015-07-12:
Assignees: @drn05r and @Tyler-Ward 
Dependencies: 
- [ ] await permission #4 
Sub Tasks:
- [ ] Source the rack so we know exact dimensions and wall mounting options.
- [ ] purchase the rack and mounting hardware
- [ ] arrange installation with susu

Tyler-Ward, 2015-07-12:
http://www.comms-express.com/products/21u-600mm-deep-data-cabinet-data-rack/ was the rack previously suggested to susu

nsefan, 2015-08-05:
The playroom has been temporarily closed, and will be reopened in what is currently break-out room 2 on the ground floor. The former playroom has been converted into an office. Need to discuss at length with SUSU to determine exact location of internal radio equipment.

Tyler-Ward, 2015-08-05:
More info on room move in #44

Tyler-Ward, 2015-08-07:
The room for the control kit has changed and installation will be carried out in a new location when a room has been decided. see #44 for more info.

closed 2015-08-07 by Tyler-Ward