SUWS/mast-planning #44: Playroom has been moved. Location of kit needs to be confirmed.

nsefan, 2015-08-05:
The playroom has been temporarily closed, and will be reopened in what is currently break-out room 2 on the ground floor by the end of August 2015. The former playroom has been converted into an office. Need to discuss at length with SUSU to determine exact location of internal radio equipment.

This will also affect SURGE's usage of the room for their transmitter.

Tyler-Ward, 2015-08-05:
These may be of use, both of these where in the last union council. @mnestis can you confirm wither these passed or not as there are no minutes yet?

https://www.susu.org/downloads/democracy/papers/9_977_1433358928.pdf

https://www.susu.org/downloads/democracy/papers/9_977_1433712397.pdf

mnestis, 2015-08-05:
The Union Films one was referred to the trustees, and I can't remember about the other one.

Tyler-Ward, 2015-08-07:
We met with Tony (Director of union services) and Kerry (VPDCI) from SUSU today 

#### The room moving plan 

The plan is to move the support staff to B40 and move the core team who need to interact with the sabs to the area around the sab offices. This will convert the entire top floor of B40 to be staff and admin areas. The playroom will be moving to the Breakout room 2 on the ground floor. Furthermore there are longer term plans to redevelop B40 whereas plans for renovating B42 shouldn't cause us any major issues.

#### Options for Us

Tony suggested considering mounting options on B42 as the plan is to make more of this building student space rather than admin. current thoughts from us are to put the Array on the white concrete wall on the right of the stairwell as you go down into the building. additional tickets will cover the work that we need to do to investigate this.

#### Timescales

Tony is next meeting with E&F is next Friday however there are more meetings planned a few weeks afterwards if we don't have a plan by then so there is no need to panic about getting it done for the next one.

Tyler-Ward, 2015-08-09:
A doc has been created to cover the options for the new site https://docs.google.com/document/d/1PL8i56SDKNnUgWACyBs5cuRdBzUVWzlg3FtfxwKsrV4